pschatzmann/arduino-audiokit #4: Is it possible to set higher sample rate I.e. 96K

y-sh1, 2022-02-17:
I am using audio kit v2.2 and tested your examples up to 48khz sampling rate. However for my project I need a higher sampling rate of 96kHz. Is this possible?

pschatzmann, 2022-02-17:
I did not test, but here are the supported values from ADF that I was mirroring in my code:

    switch (sample_rate) {
      case AUDIO_HAL_08K_SAMPLES: /*!< set to  8k samples per second */
        return 8000;
      case AUDIO_HAL_11K_SAMPLES: /*!< set to 11.025k samples per second */
        return 11000;
      case AUDIO_HAL_16K_SAMPLES: /*!< set to 16k samples in per second */
        return 16000;
      case AUDIO_HAL_22K_SAMPLES: /*!< set to 22.050k samples per second */
        return 22000;
      case AUDIO_HAL_24K_SAMPLES: /*!< set to 24k samples in per second */
        return 24000;
      case AUDIO_HAL_32K_SAMPLES: /*!< set to 32k samples in per second */
        return 32000;
      case AUDIO_HAL_44K_SAMPLES: /*!< set to 44.1k samples per second */
        return 44000;
      case AUDIO_HAL_48K_SAMPLES: /*!< set to 48k samples per second */
        return 48000;
    }

y-sh1, 2022-02-23:
Thank you Phil,
I am finally able to play @96KHz but only after setting the channel_format to I2S_CHANNEL_FMT_ONLY_RIGHT (I.e. mono). This is still good enough for me.
However, now I am facing another obstacle. For my project I need to generate audio output AND simultaneously read and process the input from a microphone. I looked at the streaming  examples in your arduino-audio-tools (which is BTW a great framework), but could not figure out how to do it.  I would highly appreciate your guidance…

pschatzmann, 2022-02-23:
Two examples:
- https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-i2s
- https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-i2s-2

I would prefer the first one...

y-sh1, 2022-02-23:
Thank you. 
Thanks.
Will it work on the audio kit board or do I need to make adjustments?

Maybe I need to clarify that audio is played from sd card and I need to get the microphone’s readings simultaneously and independently. 

So how can I access/process the microphone data in real-time?

pschatzmann, 2022-02-23:
That's easy, just call kit.readBytes(buffer, size) and do with that data whatever you need to do

closed 2022-03-09 by pschatzmann