pschatzmann/arduino-audiokit #6: Codec mode and input settings (line in)

IsiTapier, 2022-03-09:
Why is LINE_IN in the audio_hal_codec_mode_t setting and not in the audio_hal_adc_input_t setting?
Isn't it possible to have AUDIO_HAL_CODEC_MODE_BOTH activated and still LINE_IN selected as Input source? (simultaneously doing line in and line out)

pschatzmann, 2022-03-09:
These are different concepts. In audio_hal_codec_mode_t you specify if you do input or output or both and audio_hal_adc_input_t specifies the input from which device you receive the input.  

By the way audio_hal_adc_input_t does not work properly on the AudioKit...

IsiTapier, 2022-03-09:
Oh good to know
Do you know whether it is working for the esp32 lyra-t?
And does it work if you set the audio_hal_codec_mode_t to AUDIO_HAL_CODEC_MODE_LINE_IN?

pschatzmann, 2022-03-09:
I haven't tried. 
I recommend to use my [arduino-audio-tools](https://github.com/pschatzmann/arduino-audio-tools) project which uses this library and adds an easy to use layer on top of this and there are plenty of [examples](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit)...

IsiTapier, 2022-03-09:
I wonder why in the Ai-thinker config, line in is set as an input and in your config it goes as a codec mode instead.
https://github.com/pschatzmann/arduino-audiokit/blob/main/src/audio_hal/audio_hal.h:
[image]
https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/blob/master/ESP32_AC101_Driver/ai_thinker_audio_kit_v2_2/board_def.h:
[image]

closed 2022-03-09 by IsiTapier

pierluigizagaria, 2022-06-09:
In my case I have a Audio kit (ES8388) 2974 and I cannot get line in audio. LINE_1 gives buzzing, LINE_2 only the on-board microphone. What can I do to debug this thing?

pschatzmann, 2022-06-09:
https://www.pschatzmann.ch/home/2021/12/15/the-ai-thinker-audiokit-audio-input-bug/

pierluigizagaria, 2022-06-09:
It's an open question then. In my case I don't event get the line in mixed. I've used a Y adapter for a headset, and nothing gets captured from the headset microphone. 
Thank for the fast answer!

pschatzmann, 2022-06-09:
Maybe there is just not enough power ? 
I tested with an IPAD having the power 3/4 up...

pierluigizagaria, 2022-06-09:
Tried to input sound from output of my pc with a jack to jack adapter. I get sound but is very distorted. I was trying to make a voip intercom with this board but is becoming more and more frustrating.

pschatzmann, 2022-06-09:
If you need to connect the audiokit with the pc you could try to use Wifi or Bluetooth to transmit the audio...

pierluigizagaria, 2022-06-09:
Oh no is just for the audio in test you suggested. But no luck at all

pschatzmann, 2022-06-09:
Ah, so voip intercom should still be possible...

pierluigizagaria, 2022-06-09:
I was thinking about using the SIP protocol so I can think about my intercom as a phone and send a key number code to open the door. got everything working except the microphone input with esp-adf.

pschatzmann, 2022-06-09:
I haven't worked with ADF, but  I think that's strange: have you tried both AUDIO_HAL_ADC_INPUT_LINE1 and AUDIO_HAL_ADC_INPUT_LINE2 ?

pierluigizagaria, 2022-06-09:
Line 2 works only for the built in microphone. Line 1 low volume buzzing.

pschatzmann, 2022-06-09:
Is the built-in microphone not good enough ?

pierluigizagaria, 2022-06-09:
With external jack I could take the intercom audio and pipe it into the board so it can send it to my phone and vice versa with the jack out.
I could do it even without jacks and solder wires, but this board has 2 jacks and I would like to make this crappy board do what it's built for..

pschatzmann, 2022-06-09:
If you check the [comment from Uzmeyer](https://www.pschatzmann.ch/home/2021/12/15/the-ai-thinker-audiokit-audio-input-bug/): If I am correct, the only way to fix this is taking MIC2 out of the circuit, simplest by removing the coupling caps C19 and C20, then you would be left with a working stereo line-in on LIN2/RIN2 and mono microphone on LIN1+RIN1.

pierluigizagaria, 2022-06-09:
What are the coupling caps? Sorry for the stupid question!

pierluigizagaria, 2022-06-13:
I've removed the mic2 from the board and set the line 2. 
The line in with headset still doesn't work BUT mic1 captures audio if I plug the jack into the board line in.
This is driving me crazy.

pierluigizagaria, 2022-06-13:
Desoldered the second mic too. The line2 with linein attached work in a strange way. 
If I use a Y adapter and plug a headset with microphone into the board, I can hear sound but the microphone doesn't work. HOWEVER If I put the headphone plug into che linein, my headset speaker act as a microphone!
What's happening?

pschatzmann, 2022-06-13:
Are you using this kind of adapter ? https://www.digitec.ch/de/s1/product/inline-headset-adapter-015-m-einstiegsklasse-35mm-klinke-aux-audio-kabel-13348379?dbq=1&gclid=Cj0KCQjwwJuVBhCAARIsAOPwGAQ-8ttNQ2S3Vav528iSiG3mg4sS6qnpAja8LrXsTdyM5azyC0Kq2_YaAvtKEALw_wcB&gclsrc=aw.ds

pschatzmann, 2022-06-13:
I tested with some earphones and when I plug them into Linein like you did, I don't get any output!
So, either your board (version) is messed up, or then signs on your headset are wrong...

pschatzmann, 2022-06-13:
My gut feeling is that the output of the microphone is just not strong enough to be picked up. Can you test with another audio source. E.g. connect your mobile phone, play music and use full volume

pierluigizagaria, 2022-06-13:
If I connect the headphone to the line in, the headset speakers act as a microphone and works!
It should not be a power problem!