pschatzmann/arduino-audiokit #7: Troubles with getting aux in

IsiTapier, 2022-03-12:
I have another question, 
yesterday my esp32 lyrat board arrived and i tried read out line in with the streams-audiokit-webserver_wav example from your Audio tools library. Unfortunately I dont't receive anything from the aux in line, only the microphones. I tried setting `result.codec_mode` to `AUDIO_HAL_CODEC_MODE_LINE_IN` at AudioKit.h line 37. This lead to the aux in getting send to the headphone output (without any sound of the microphones, which is good at least), but after a while I get a lot of these errors:
```
E (59218) I2S: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1352 (i2s_read):rx NULL
Error:   i2s_read
```
and the audio aswell isn't received from the esp.

Do you maybe have an idea what I could do to fix this or what other approches could work? I did also tried to search through the code auf the audio hal system but I couldn't find the part where the codec_mode gets passed to the module. Maybe you know where this happens, then it might be possible to adjust it so that the audio comes through like when I set it to line_in mode and rx working aswell. Maybe initializing aux in line manually or something like that.
My overall goal is to read out the aux in and stream it on wifi. (If there is another way of doing this)

Thanks for your awesome work in your libraries and your quick respondes and help.

pschatzmann, 2022-03-12:
I suggest you have a look at the examples which read audio (e.g. [this](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-serial/streams-audiokit-serial.ino).
There you see how to specify the input device....

IsiTapier, 2022-03-12:
Thanks a lot, you savaed me a lot of time trying useless tings, it was a lot easy then assumed.
It seems like LINE_1 are the microphones and LINE_2 is the aux in. (LINE_IN mode is probably internally sending the line_in directly to the outputs without sending them to the esp)
Maybe it would reduce confusion to clearly mark this in code, I found the description `mic input to adc channel 2` and the name `AUDIO_HAL_CODEC_MODE_LINE_IN` at audio_hal.h (line 66, 73, 74) especially confusing. (Shall I do a pull request for that)

A few more little things:
1. Do you know why after uploading to the board, it first continues executing the old code, untill I press RST?
2. Is it possible to reduce the background noise? (When I stop my audio source I hear some)
3. When uploading it allways doing the old `Connecting......_____......_____.......` stuff, and only stops after pressing BOOT + RST, is there a way to prevent this?

closed 2022-03-31 by pschatzmann