pschatzmann/arduino-audiokit #8: Weired hardware revision of Esp32-audio-kit

fricklr, 2022-03-29:
Hi,
thanks for that huge amount of genius work! 
I just checked some examples and came to a point where I need some help.
I have got some Esp32-audio-kit boards which are labeled with revision 2957 but I only can get them to work and make sound if I select the AC101 option 6.
When I listen to a url in [player-url_icy-audiokit](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/player-url_icy-audiokit) the sound is good (I had to comment out kit.processActions(); in the loop otherwise it would halt playing for a brief moment every second -- [I] AudioKit.h : 352 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*) is somehow triggered). 
In other examples the sound is heavily distorted but I hear something.
My main problem is I can't get the SD-card to be detected. I got it to work previously with the SD_MMC lib and I checked different boards, so I don't think it is a hardware issue.
That's the error from sdbegin (I tried options for boards 5 - 7 all having the same error):

[E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
[E][sd_diskio.cpp:775] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
Card Mount Failed

I use PlatformIO with freshly installed libs from github (today)
Do you have any idea how I could get it to work?
Thanks in advance!

platformio.ini
```
[env:esp32dev]
platform = espressif32
board = esp32dev
lib_ldf_mode = deep+
lib_deps = greiman/SdFat
           https://github.com/pschatzmann/arduino-audio-tools.git
           https://github.com/pschatzmann/arduino-libhelix.git
           https://github.com/pschatzmann/arduino-audiokit.git
           https://github.com/pschatzmann/arduino-midi
framework = arduino
build_flags = -DAUDIOKIT_BOARD=6 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args 
```

pschatzmann, 2022-03-30:
I suggest that you check the dip switches: I think the best setting is OFF, ON, ON, OFF, OFF

fricklr, 2022-03-30:
Thanks for the fast response! I think I'm almost there. SD is working now - I totally overlooked these little creatures ;) Your audio.mp3 also works, but another one from me did not (I assume it's a codec thing - will check that later). 
Thanks again - you made my day :)

closed 2022-03-30 by fricklr