pschatzmann/arduino-audiokit #9: Audiokit v2.2 (2957), version with I2C on 32/33 - two suggestions

perlix, 2022-03-31:
After I got my 'flavour' of the infamous Audiokit v2.2 (2957) with ES8388 to work, I have two suggestions that may shorten the route for other beginners:

1. Perhaps this version of the board could be added as something like '8) ai_thinker (ES8388) 2957 (I2C on 32/33)' to the AudioKitSettings.h file, and then made selectable via `#define AUDIOKIT_BOARD 8`, with the following pin definitions in its board_pins_config.c file:
```
	i2c_config->sda_io_num = GPIO_NUM_33; 
	i2c_config->scl_io_num = GPIO_NUM_32;
	i2s_config->bck_io_num = GPIO_NUM_27;
	i2s_config->ws_io_num = GPIO_NUM_25;
	i2s_config->data_out_num = GPIO_NUM_26;
	i2s_config->data_in_num = GPIO_NUM_35;
```
Unlike the 18/23 version, this board has no conflict with GPIO18, so volume control through KEY5 and KEY6 will work.

2. On this flavour of the v2.2 (ES8388) 2957 board, the current file es8388.c results in very low max volume on speakers as well as headphones. An elsewhere suggested fix that worked for me was changing function _es8388_get_voice_volume()_ in that file as follows:

```
esp_err_t es8388_set_voice_volume(int volume) {
  esp_err_t res = ESP_OK;
  if (volume < 0) volume = 0;
  else if (volume > 100) volume = 100;
  volume /= 3;
  res = es_write_reg(ES8388_ADDR, ES8388_DACCONTROL4, volume);
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL5, volume);
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL24, volume);
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL25, volume);
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL26, volume);
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL27, volume);
  return res;
}
```
So perhaps this board could have its own _es8388_set_voice_volume()_ function?

pschatzmann, 2022-03-31:
Hmm - the pins are the same like 5) ai_thinker (ES8388) 3478.
Did you try with that option ?

perlix, 2022-03-31:
Thanks again for the ultrasonic response! 
You're right (of course), just tried option 5 and that works fine for my 2957, so please forget my first suggestion. 

You'll probably come up with a better alternative for my second suggestion as well, but I was so pleased to have found it after days of tinkering that I wanted to share it, just in case...

Thanks for your flawless and well documented libraries!

pschatzmann, 2022-03-31:
I will update the Readme and add the 2957 to option 5 as well.
For the issue with the volume I don't have an easy solution. Was the issue related to earphones or to the speakers ?

pschatzmann, 2022-03-31:
For the volume I added an additional option in the AudioKitSettings.h. Can you help me to test this:
#define AI_THINKER_ES8388_VOLUME_HACK 1

perlix, 2022-04-01:
Just installed the newest version and tested a clean _player-url_icy-audiokit_ example from the _arduino-audio-tools_ library with these two lines added atop:
```
#define AUDIOKIT_BOARD 5 
#define AI_THINKER_ES8388_VOLUME_HACK 1
```
To my surprise, max volume is very low again. It appears that the unmodified es8388_set_voice_volume() was compiled, despite the new `#if AI_THINKER_ES8388_VOLUME_HACK` directive in es8388.c. Changing it to `#ifdef AI_THINKER_ES8388_VOLUME_HACK` makes everything work perfectly (on my kit). Beats me, but I'm on highly unfamiliar ground here.

(Hope I'm not bothering you too much with my strange version of this kit).

pschatzmann, 2022-04-01:
I confirm adding this in the sketch is not good enough. You need to update the AudioKitSettings.h file

perlix, 2022-04-01:
Moved ` #define AI_THINKER_ES8388_VOLUME_HACK 1` from the sketch to _AudioKitSettings.h_ and everything now works like the Swiss public transport! As I said: unfamiliar ground for me, so thanks for your patience.

pschatzmann, 2022-04-01:
Cool

closed 2022-04-01 by pschatzmann