pschatzmann/arduino-audiokit #12: ESP32 S2 - Not compiling

jay6621, 2022-04-05:
Thanks for Both of your audio libraries, I came here after listening to a guy with the swiss accent ;)

I am trying to run "streams-audiokit-serial" , but it is not compiling and throws below error.

```
C:\Users\User\OneDrive\Documents\Arduino\libraries\arduino-audiokit-main\src/AudioKitHAL.h:27:10: fatal error: esp_a2dp_api.h: No such file or directory
 #include "esp_a2dp_api.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
```

My question is did you managed to compile this for any ESP 32 S2 board ?

Thanks and Regards 
JAY

pschatzmann, 2022-04-05:
This does not make any sense: all AudioKits use an ESP32 and to my knowledge there is no AudioKit board with S2.
If you have a S2 board I suggest to use the Streams examples. Please note however that, since I do not have any S2 board, I could never test this...

pschatzmann, 2022-04-05:
I committed some corrections

jay6621, 2022-04-06:
Whoa... Thank you for quick response. 
I can confirm that audiokits examples "streams-audiokit-webserver_wav" and "streams-audiokit-serial" 
from arduino-audio-tools library is working as expected in Kaluga v1.2 board (it has ESP32-S2).
I have Selected "ESP32S2 DEV MODULE" as board in Arduino IDE and
output for "streams-audiokit-webserver_wav" is as below:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x570
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28d8
entry 0x4004c18c
starting AudioKit...
[W] AudioKit.h : 43 - sd_active = false
[W] AudioKit.h : 543 - Deactivating SPI because SD is not active
[W] AudioKit.h : 534 - Undefined ActionLogic for pin: 4 
[W] AudioActions.h : 72 - pin -1 -> Ignored
[W] AudioKit.h : 534 - Undefined ActionLogic for pin: 1 
[W] AudioKit.h : 534 - Undefined ActionLogic for pin: 0 
[W] AudioActions.h : 72 - pin 0 -> Ignored
AudioKit started
......
IP address: 192.168.0.104
```
Audio is streaming perfectly. (roughly 4 second lag)
Thank You So much for your time.

closed 2022-04-07 by jay6621