pschatzmann/arduino-audiokit #18: output.ino example stops working after a few minutes

vanniaz, 2022-05-03:
I am encountering a strange behaviour with the output.ino example (on a LyraT board): the sinewave runs perfectly for a couple of minutes, then it suddenly changes its frequency to a lower note, and later it begins stuttering.
Is there something that I should check? This is the output of the serial monitor:

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Info:    bool AudioKit::begin(AudioKitConfig)
Info:    Selected board: 1
Info:    headphone detection is active
Info:    i2c sda: 18
Info:    i2c scl: 23
Info:    i2c clk_speed: 100000
Info:    audio_codec_initialize-END-OK
Info:    i2s_set_pin
Info:    I2S0, MCLK output by GPIO0
Info:    Codec mode is 2, Ctrl:1

pschatzmann, 2022-05-03:
This is just a primitive implementation of a sine generator. If you want to have a stable signal you need to use the SineWaveGenerator or SineFromTable from the [audio-tools](https://github.com/pschatzmann/arduino-audio-tools)

vanniaz, 2022-05-03:
Thanks, I will try it.

vanniaz, 2022-05-03:
I did a quick test by simply taking the waveform generation out of the I2S write loop and filling the buffer with exactly 4 sine periods:
```
#include <Arduino.h>
#include <Wire.h>
#include "AudioKitHAL.h"

AudioKit kit;

const int BUFFER_SIZE = 1024;

const int N_SAMPLES = BUFFER_SIZE / (sizeof(uint16_t) * 2);

uint8_t buffer[BUFFER_SIZE];

void setup() {
  LOGLEVEL_AUDIOKIT = AudioKitInfo; 
  Serial.begin(115200);
  // open in write mode
  auto cfg = kit.defaultConfig(AudioOutput);
  kit.begin(cfg);

float angle = 0;

 int16_t *ptr = (int16_t*)buffer;

  for ( int i = 0; i < N_SAMPLES; i++)
  {
    angle = (float) i * 4 * 2 * PI / N_SAMPLES;
    int16_t result = 32767.0 * sin(angle);
    *ptr++ = result;
    *ptr++ = result;
  }
}

void loop() {
  kit.write(buffer, BUFFER_SIZE);
}
```
I confirm that it keeps running, so the problem was definitely in the sinewave generation, not in the kit.write part (which is a good thing, since that is the most important part!).
I am closing this issue.

closed 2022-05-03 by vanniaz