pschatzmann/arduino-audiokit #19: SD card via SDMMC instead of SPI?

perlix, 2022-05-07:
Grüezi Phil,

Triggered by the wiki page about SPI and I2C, I wondered if you ever considered to access the SD card over the fast SDMMC bus instead of SPI. The Audiokit datasheet suggests that the SD slot is connected to all 6 pins of ESP32's SDMMC_HOST_SLOT_1 (2, 4, 12, 13, 14 and 15, all with pullup resistors).

Using the SD_MMC library, I verified that SDMMC 1-bit mode (using GPIO 2, 14 and 15) is almost twice as fast as SPI.
(Couldn't get the even slightly faster SDMMC 4-bit mode working, so GPIO 4 and/or 12 may not actually be connected on my kit).

This is just an idea. I don't even know if SD speed could ever become a bottleneck for audio projects, but if a board's internal wiring allows it, I always prefer using the SDMMC bus.

pschatzmann, 2022-05-07:
Yes, I confirm that this is working and nothing is stopping you to use this: Maybe I should add a chapter in the Wiki.
 
I developed this project as an 'add on' to my AudioTools library for which I prefer solutions that __work across different microcontrollers__.

perlix, 2022-05-07:
Thanks, that makes sense.

SDMMC pins are fixed (cannot be remapped). For 1-bit mode:
2	DATA0
14	CLK
15	CMD

So I changed your sdbegin.ino example to:

```
#include "AudioKitHAL.h"
#include "SD_MMC.h"

AudioKit kit;

void setup(){
    pinMode(2, INPUT_PULLUP);
    pinMode(15, INPUT_PULLUP);
    Serial.begin(115200);
    if(!SD_MMC.begin("/sdcard", true)){    // 1-bit mode
        Serial.println("Card Mount Failed");
        return;
    } else {
        Serial.println("Card Mount Success");
    }
}

void loop() {

}
```

pschatzmann, 2022-05-07:
I thught thats what the following methods are for:
 -   bool setPins(int clk, int cmd, int d0);
 -   bool setPins(int clk, int cmd, int d0, int d1, int d2, int d3);

perlix, 2022-05-07:
Correct. SD_MMC.begin doesn't accept pin specifications anyway (because they're fixed).

closed 2022-05-12 by pschatzmann