pschatzmann/arduino-audiokit #22: Which board to select in Arduino IDE

rjrajbir, 2022-05-19:
Hello there,
I have[ ESP32 Audio Devkit](https://robu.in/product/esp32-audio-kit/) and I have downloaded this code but I am not able to compile this code, could you please let me which board should I select or how to add this board in board manager.

Best Regards
Rajbir

rjrajbir, 2022-05-19:
I have shared the link

rjrajbir, 2022-05-19:
[image]

pschatzmann, 2022-05-19:
A149 ? This is different then your original link.
I have never seen this before. I would try with option 5 first

rjrajbir, 2022-05-20:
Alright, thank you for your help but I am asking for arduino IDE
[image]

pschatzmann, 2022-05-20:
In Frequently Ask Questions you can find:
Question: Which Board do I need to select?
Answer: If the board is based on an ESP32, use the ESP32 Dev Module. Only the esp32_s2_kaluga_1_v1_2 is based on an ESP32-S2!

rjrajbir, 2022-05-21:
Ok, but it still gives the error
[image]

pschatzmann, 2022-05-21:
Your confusing me: I close this issue because it has been answered and you did not respond on your other issue

closed 2022-05-21 by pschatzmann