pschatzmann/arduino-audiokit #23: Showing SineWaveGenerator.h:45:20: error: 'result' is used uninitialized in this function [-Werror=uninitialized]              size_t result;

rjrajbir, 2022-05-20:
[image]

Showing type define error.

Please solve this issue.

BR
Rajbir

pschatzmann, 2022-05-20:
Hmm, I can't reproduce you issue. 
Are you sure that you have selected ESP32 Dev Module as board ?
If you go to -> Tools -> Board -> Board Manager and Enter ESP32. What version are you using?
If you use the actual version you should see esp32 by Espressif Systems version 2.0.3

rjrajbir, 2022-05-21:
[image]
This the board I am using.
[image]
My latest board version is 1.0.6

ok, thank you for your quick response.

rjrajbir, 2022-05-21:
Yes, you were right, my compiler warning was set to ALL, by setting it default it worked.
Thank you so much

closed 2022-05-21 by rjrajbir

pschatzmann, 2022-05-21:
Still strange: on my systems I don't get any error even if the compiler setting is set to ALL...