pschatzmann/arduino-audiokit #24: esp32-s3-korvo-v3

Tatvatecnovation, 2022-05-25:
Hi Sir ;

I have a esp32s3-korvo-v3 board .
I am working for bidirectional audio live streaming on http.
I got a following issue on streams_audiokit_webserver_acc example .

C:\Users\tatva\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioAnalog/AnalogAudioESP32.h:7:10: fatal error: soc/dac_channel.h: No such file or directory
 #include "soc/dac_channel.h"

pschatzmann, 2022-05-25:
I don't get this error and I assume that you use an old version of the library.
However the major problem will be that there is no board support for the korvo s3 board in the AudioKit project and you would need to extend this library yourself....

Tatvatecnovation, 2022-05-25:
Thank you for Response.
I get the point : Korvo S3 is not supported. Ok than I would like to extend this beautiful library.
Any Hint on How can I Do that..?
If I make Progress I will post it here, So everyone can get benefit.
Thank You.

pschatzmann, 2022-05-25:
1. You would clone this library first and do the changes there. 
2. Follow https://github.com/pschatzmann/arduino-audiokit/wiki/Adding-Support-for-a-new-Board
3. After it is working you can submit a pull request.

Tatvatecnovation, 2022-05-25:
Thank You for quick response. 
I am on it Now. I will post the update here.

Tatvatecnovation, 2022-05-25:
ok.. So, I managed to compile and upload [streams-audiokit-webserver-wav](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/streams-audiokit-webserver_wav) on Korvo 2 S3.
No Audio on webserver with below Errors:
```
starting AudioKit...
[W] AudioKit.h : 43 - sd_active = false
[W] AudioKit.h : 557 - Deactivating SPI because SD is not active
[W] AudioKit.h : 548 - Undefined ActionLogic for pin: 4 
[W] AudioActions.h : 73 - pin -1 -> Ignored
[W] AudioKit.h : 548 - Undefined ActionLogic for pin: 1 
[W] AudioKit.h : 548 - Undefined ActionLogic for pin: 0 
[W] AudioActions.h : 73 - pin 0 -> Ignored
AudioKit started
.........
IP address: 192.168.250.112
[E] AudioCopy.h : 253 - write to target has failed!
```

pschatzmann, 2022-05-25:
Difficult to say what the issue is: I suggest to test the following functions separately
- sd drive
- output
- input

you can use the example sketches in the project

Tatvatecnovation, 2022-05-25:
I am working on separate tests, here's my changed version of this library as per your suggestions : [My Fork](https://github.com/Tatvatecnovation/arduino-audiokit)
I will update the progress here.

Thank you.

Tatvatecnovation, 2022-05-28:
Hi.. I have tested separate function as per your suggestion.
1. output (example from this library)
got the below log without any voice from speaker.
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
SHA-256 comparison failed:
Calculated: 27beaa143e4753b6cdeed5b388d3ebbf4c921ce01156fd10673139a323a25557
Expected: 6eb14ec480aff4f05e39378c41406102ac163573741e63b6cfd73509ef44e3bd
Attempting to boot anyway...
entry 0x403b61d8
Info:    bool AudioKit::begin(AudioKitConfig)
Info:    Selected board: 11
Info:    headphone detection not supported
Info:    i2c sda: 17
Info:    i2c scl: 18
Info:    i2c clk_speed: 100000
Info:    ES8311 in Slave mode
Info:    audio_codec_initialize-END-OK
Info:    i2s_set_pin
Info:    Codec mode is 2, Ctrl:1

```

2. input (example from this library)
got the below log, with 0, 0 values.
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
SHA-256 comparison failed:
Calculated: 27beaa143e4753b6cdeed5b388d3ebbf4c921ce01156fd10673139a323a25557
Expected: 6eb14ec480aff4f05e39378c41406102ac163573741e63b6cfd73509ef44e3bd
Attempting to boot anyway...
entry 0x403b61d8
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
0, 0
```

3. sd_begin (example from this library)
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
SHA-256 comparison failed:
Calculated: 27beaa143e4753b6cdeed5b388d3ebbf4c921ce01156fd10673139a323a25557
Expected: 6eb14ec480aff4f05e39378c41406102ac163573741e63b6cfd73509ef44e3bd
Attempting to boot anyway...
entry 0x403b61d8
Card Mount Failed

```
I am still trying to find the flow of library, I am pretty sure we can use IDF (or ADF) Header and C files with some modifications.
Hope to hear your view on this.
Thank you.

pschatzmann, 2022-05-28:
The only issue that I could identify is that the i2s master clock pin is not set up. 
I suggest to test with commenting out  line 143 which sets result.mck_io_num = I2S_PIN_NO_CHANGE in AudioKitHAL.h.
It seems that on this board this should be GPIO16...

Tatvatecnovation, 2022-05-29:
> The only issue that I could identify is that the i2s master clock (pin) is not set up.

after commenting `result.mck_io_num = I2S_PIN_NO_CHANGE` 
**out** example is working fine and I got Sine wave sound from speaker.
Thank you for that.

I am still struggling to get [streams-url_mp3-audiokit](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/streams-url_mp3-audiokit) running.

as per your suggestions of adding log lines, I got the idea that for S3 I need to add support in [AudioConfig.h](https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioConfig.h) in arduino-audio-tools library, So, I changed line 144 of [AudioConfig.h](https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioConfig.h) from `#if defined(ESP32)  && !defined(ARDUINO_ESP32C3_DEV)`  to
`#if defined(ESP32)  && !defined(ARDUINO_ESP32C3_DEV) && !defined(ARDUINO_ESP32S3_DEV)` and tried to add support for S3 by adding below snippet (above ESP32 support part)

```
//-------ESP32 S3---------
#if defined(ESP32)  && defined(ARDUINO_ESP32S3_DEV) 
#include "esp32-hal-log.h"
// optional libraries
//#define USE_A2DP
//#define USE_ESP8266_AUDIO

#define USE_PWM
#define USE_URL_ARDUINO
#define USE_I2S
#define USE_AUDIO_SERVER
#define USE_TYPETRAITS
#define USE_EFFECTS_SUITE
#define USE_TIMER

#define PWM_FREQENCY 30000
#define PIN_PWM_START 12
#define PIN_I2S_BCK 9
#define PIN_I2S_WS 45
#define PIN_I2S_DATA_IN 10
#define PIN_I2S_DATA_OUT 
#define I2S_USE_APLL false  
// Default Setting: The mute pin can be switched off by setting it to -1. Or you could drive the LED by assigning LED_BUILTIN
#define PIN_I2S_MUTE 23
#define SOFT_MUTE_VALUE LOW  
#define PIN_CS SS
#define PIN_ADC1 34 
#define PIN_ADC2 14

#define I2S_AUTO_CLEAR true

// URLStream
//#define USE_URLSTREAM_TASK
#define URL_STREAM_CORE 0
#define URL_STREAM_PRIORITY 2
#define URL_STREAM_BUFFER_COUNT 10
#define STACK_SIZE 30000
#define URL_CLIENT_TIMEOUT 60000;
#define URL_HANDSHAKE_TIMEOUT 120000

// Default LED
#ifndef LED_BUILTIN
# define LED_BUILTIN 13 // pin number is specific to your esp32 board
#endif

// support for old idf releases
#if ESP_IDF_VERSION_MAJOR < 4 && !defined(I2S_COMM_FORMAT_STAND_I2S)
# define I2S_COMM_FORMAT_STAND_I2S (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB)
# define I2S_COMM_FORMAT_STAND_MSB (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB)
# define I2S_COMM_FORMAT_STAND_PCM_LONG (I2S_COMM_FORMAT_PCM | I2S_COMM_FORMAT_PCM_LONG)
# define I2S_COMM_FORMAT_STAND_PCM_SHORT (I2S_COMM_FORMAT_PCM | I2S_COMM_FORMAT_PCM_SHORT)

typedef int eps32_i2s_sample_rate_type;
#else
typedef uint32_t eps32_i2s_sample_rate_type;
#endif

#endif
[... 195 lines cut ...]
Debug:   audio_free
[I] AudioOutput.h : 720 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 733 - FormatConverterStream: no conversion
[W] AudioKit.h : 43 - sd_active = true
Info:    bool AudioKit::begin(AudioKitConfig)
Info:    Selected board: 11
Info:    headphone detection not supported
Info:    i2s port configured!!!!
Info:    i2s_set_pin
Info:    ESP32_S3_KORVO_2
Debug:   audio_hal_ctrl_codec

assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))

Backtrace:0x40377482:0x3fce2b900x4037c8b1:0x3fce2bb0 0x40382489:0x3fce2bd0 0x4037d846:0x3fce2d00 0x420101d5:0x3fce2d40 0x42010113:0x3fce2d60 0x42006055:0x3fce2d80 0x42006695:0x3fce2df0 0x420066a5:0x3fce2e70 0x420045b1:0x3fce2e90 0x42005dcb:0x3fce2ed0 0x42005e23:0x3fce2ef0 0x42005c9a:0x3fce2f50 0x42005d1e:0x3fce2f80 0x42008939:0x3fce2fa0 0x42003e85:0x3fce2fc0 0x42007364:0x3fce2fe0 0x4200748b:0x3fce3000 0x4200751e:0x3fce3040 0x42011e19:0x3fce3060 

ELF file SHA256: 0000000000000000

Rebooting...
```
I think I got the I2S pins right.
Can you guide what else do I need to change?
Pardon me if I am stupidly wrong. I am still studying the Library.

Thank You.

pschatzmann, 2022-05-29:
It seems that the initialization 
Info:    bool AudioKit::begin(AudioKitConfig)
Info:    Selected board: 11
is running twice and the board does not like this!

pschatzmann, 2022-06-01:
You really need to create a stacktrace to analyse where this is coming from.
https://github.com/me-no-dev/EspExceptionDecoder

Tatvatecnovation, 2022-06-01:
I tried me-no-dev's EspExceptionDecoder
and got below result in decoder window.
```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x42006061: AudioKit::begin(AudioKitConfig) at C:\Users\AJ\Documents\Arduino\libraries\arduino-audiokit\src/AudioKitHAL.h line 320
0x420066a1: audio_tools::AudioKitStream::setAudioInfo(audio_tools::AudioBaseInfo) at C:\Users\AJ\Documents\Arduino\libraries\arduino-audio-tools\src/AudioLibs/AudioKit.h line 265
0x420045bd: audio_tools::MP3DecoderHelix::infoCallback(_MP3FrameInfo&) at C:\Users\AJ\Documents\Arduino\libraries\arduino-audio-tools\src/AudioCodecs/CodecMP3Helix.h line 136
0x42005dd7: libhelix::MP3DecoderHelix::provideResult(_MP3FrameInfo&) at C:\Users\AJ\Documents\Arduino\libraries\arduino-libhelix\src/MP3DecoderHelix.h line 165
0x42005e2f: libhelix::MP3DecoderHelix::decode(libhelix::Range) at C:\Users\AJ\Documents\Arduino\libraries\arduino-libhelix\src/MP3DecoderHelix.h line 123
0x42005ca6: libhelix::CommonHelix::writeFrame(void const*, unsigned int) at C:\Users\AJ\Documents\Arduino\libraries\arduino-libhelix\src/CommonHelix.h line 225
0x42005d2a: libhelix::CommonHelix::write(void const*, unsigned int) at C:\Users\AJ\Documents\Arduino\libraries\arduino-libhelix\src/CommonHelix.h line 124
0x42008945: audio_tools::MP3DecoderHelix::write(void const*, unsigned int) at C:\Users\AJ\Documents\Arduino\libraries\arduino-audio-tools\src/AudioCodecs/CodecMP3Helix.h line 109
0x42003e91: audio_tools::EncodedAudioStream::write(unsigned char const*, unsigned int) at C:\Users\AJ\Documents\Arduino\libraries\arduino-audio-tools\src/AudioCodecs/AudioEncoded.h line 328
0x42007370: audio_tools::StreamCopyT ::write(unsigned int, unsigned int&) at C:\Users\AJ\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/AudioCopy.h line 247
0x42007497: audio_tools::StreamCopyT ::copy() at C:\Users\AJ\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/AudioCopy.h line 267
0x4200752a: loop() at C:\Users\AJ\Documents\Arduino\libraries\arduino-audio-tools\src/AudioTools/AudioCopy.h line 337
0x42011df9: loopTask(void*) at C:\Users\AJ\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32\main.cpp line 50
```

pschatzmann, 2022-06-01:
Now you need to check in your AudioKitHAL.h line 320 what could cause the nullpointer exception...

pschatzmann, 2022-06-01:
hmm are you sure ? I can't see what could go wrong here

Tatvatecnovation, 2022-06-01:
Yes, I am certain sir, also you have already pointed this thing in this [comment](https://github.com/pschatzmann/arduino-audiokit/issues/24#issuecomment-1140498530)

Tatvatecnovation, 2022-06-02:
Hi, sir I have a question : what is `fixed_mclk` used for ? It is defined at line [87](https://github.com/pschatzmann/arduino-audio-tools/blob/636a546f7aeb61c0ddeb41587b87d15819371d4a/src/AudioI2S/I2SConfig.h#L87) at **I2SConfig.h** in [arduino-audio-tools](https://github.com/pschatzmann/arduino-audio-tools).

Tatvatecnovation, 2022-06-02:
ok.. As far as I understood, if we compile any audiokit related example from [arduino-audio-tools](https://github.com/pschatzmann/arduino-audio-tools), it will get the board configuration from this ([arduino-audiokit](https://github.com/pschatzmann/arduino-audiokit)) library, right sir ?

pschatzmann, 2022-06-02:
Yes, that's correct.

Tatvatecnovation, 2022-06-02:
Setting the initial frequency 16k, 32k and 48k all gave me same result i.e. core 1 panic'ed.

> system detects that the initial sampling rate is not correct and stops and restarts with the correct sampling rate.

can you sir please point out the part of the code, which does this check ?

pschatzmann, 2022-06-02:
I think it is in AudioKit.setAudioInfo()

Tatvatecnovation, 2022-06-04:
> I think it is in AudioKit.setAudioInfo()

yes, it is in arduino-audio-tools -> AudioKit.h -> at [249](https://github.com/pschatzmann/arduino-audio-tools/blob/636a546f7aeb61c0ddeb41587b87d15819371d4a/src/AudioLibs/AudioKit.h#L249)
I commented that check part and it worked fine, of course at cost.  
Inserted a local radio station link and It is playing fine, Funny part is It is playing so normal that I didn't bother to unplug the board, than I realized that it is not my regular Bluetooth speaker and unplugged it. **Thank you so much sir**, for your Beautiful Audio Libraries. 
Serial Log is below (stream copy log is turned off)
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
SHA-256 comparison failed:
Calculated: 27beaa143e4753b6cdeed5b388d3ebbf4c921ce01156fd10673139a323a25557
Expected: 6eb14ec480aff4f05e39378c41406102ac163573741e63b6cfd73509ef44e3bd
Attempting to boot anyway...
entry 0x403b61d8
[I] AudioTypes.h : 48 - sample_rate: 44100
[I] AudioTypes.h : 49 - channels: 2
[I] AudioTypes.h : 50 - bits_per_sample: 16
[I] I2SConfig.h : 95 - rx/tx mode: TX
[I] I2SConfig.h : 96 - port_no: 0
[I] I2SConfig.h : 97 - is_master: Master
[I] I2SConfig.h : 98 - sample rate: 44100
[I] I2SConfig.h : 99 - bits per sample: 16
[I] I2SConfig.h : 100 - number of channels: 2
[I] I2SConfig.h : 101 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 103 - auto_clear:0
[I] I2SConfig.h : 110 - buffer_count:10
[I] I2SConfig.h : 111 - buffer_size:1024
[I] I2SConfig.h : 115 - pin_bck: 9
[I] I2SConfig.h : 116 - pin_ws: 45
[I] I2SConfig.h : 117 - pin_data: 8
[I] I2SConfig.h : 118 - pin_mck: 16
[W] AudioKit.h : 43 - sd_active = false
[I] AudioKit.h : 194 - Channels 2->2
[I] AudioOutput.h : 720 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 733 - FormatConverterStream: no conversion
[I] AudioEncoded.h : 248 - virtual void audio_tools::EncodedAudioStream::setNotifyAudioChange(audio_tools::AudioBaseInfoDependent&)
[I] URLStream.h : 78 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): http://103.16.47.70:9222/;
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->http://103.16.47.70:9222/;
[I] Url.h : 110 - host->103.16.47.70
[I] Url.h : 111 - protocol->http
[I] Url.h : 112 - path->/;
[I] Url.h : 113 - port->9222
.....
[I] URLStream.h : 253 - WiFiClient
[I] HttpRequest.h : 193 - process connecting to host 103.16.47.70 port 9222
[I] HttpRequest.h : 181 - connected 1 timeout 60000
[I] HttpRequest.h : 204 - Free heap: 270708
[I] HttpHeader.h : 357 - HttpRequestHeader::setValues - path: /;
[I] HttpHeader.h : 254 - HttpHeader::write
[I] HttpHeader.h : 366 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 381 - -> GET /; HTTP/1.1
[I] HttpHeader.h : 191 -  -> Host: 103.16.47.70 
[I] HttpHeader.h : 191 -  -> Connection: close 
[I] HttpHeader.h : 191 -  -> Accept: audio/mp3 
[I] HttpHeader.h : 297 -  -> <CR LF> 
[I] HttpRequest.h : 227 - Request written ... waiting for reply
[I] HttpHeader.h : 224 - HttpHeader::read
[W] HttpHeader.h : 231 - Waiting for data...
[I] HttpHeader.h : 161 - HttpHeader::readLine -> ICY 200 OK
[I] HttpHeader.h : 161 - HttpHeader::readLine -> icy-notice1:<BR>This stream requires <a href="http://www.winamp.com/">Winamp</a><BR>
[I] HttpHeader.h : 161 - HttpHeader::readLine -> icy-notice2:SHOUTcast Distributed Network Audio Server/posix(linux x64) v2.0.0.29<BR>
[I] HttpHeader.h : 161 - HttpHeader::readLine -> icy-name:Bolly Dance Hits
[I] HttpHeader.h : 161 - HttpHeader::readLine -> icy-genre:Misc
[I] HttpHeader.h : 161 - HttpHeader::readLine -> icy-url:http://192.168.60.23:9222/listen.pls
[I] HttpHeader.h : 161 - HttpHeader::readLine -> content-type:audio/mpeg
[I] HttpHeader.h : 161 - HttpHeader::readLine -> icy-pub:0
[I] HttpHeader.h : 161 - HttpHeader::readLine -> icy-br:96
[I] HttpHeader.h : 161 - HttpHeader::readLine -> 
[W] HttpRequest.h : 153 - no CONTENT_LENGTH found in reply
[I] URLStream.h : 93 - size: 0
[I] AudioKit.h : 250 - virtual void audio_tools::AudioKitStream::setAudioInfo(audio_tools::AudioBaseInfo)

```
I am playing with settings to find why check part is not working and than will move on to ES7210 (Mic input part).
Updating here asap.
Much Love and Respect for your work and Time.

pschatzmann, 2022-06-07:
I confirm, It is not used...

Tatvatecnovation, 2022-06-07:
ok, got it. Update is I can drive the ES7210 (ADC) too with this library, I can see Mic input on serial terminal using streams-audiokit-serial but with a workaround as below in [audiokit_board.h](https://github.com/pschatzmann/arduino-audiokit/blob/main/src/audio_hal/audiokit_board.h)  
```
#ifdef USE_MICROPHONE
#define AUDIO_DRIVER AUDIO_CODEC_ES7210_DEFAULT_HANDLE
#else
#define AUDIO_DRIVER AUDIO_CODEC_ES8311_DEFAULT_HANDLE
#endif
```
So, now problem is either I can use DAC chip or ADC chip. I would love to hear your view on this.

lijianyu1985, 2022-08-20:
@Tatvatecnovation I tried your branch and got error, did you see any error like this:
[image]

pschatzmann, 2022-08-20:
There is no support for the korvo s3 board in the [AudioKit](https://github.com/pschatzmann/arduino-audiokit) project

lijianyu1985, 2022-08-20:
> There is no support for the korvo s3 board in the [AudioKit](https://github.com/pschatzmann/arduino-audiokit) project

I know that, I am using @Tatvatecnovation 's fork, his fork support korvo s3

pschatzmann, 2022-08-20:
Oh, I see and unfortunately he has deactivated the issues and discussion tabs on his fork...

lijianyu1985, 2022-08-21:
> Oh, I see and unfortunately he has deactivated the issues and discussion tabs on his fork...

Yeah, that's the reason why I am here. :)