pschatzmann/arduino-audiokit #31: How to select a single MP3 file by name

merugear, 2022-07-13:
Hi there! I am looking for an implementation to have an MP3 played by the press of a button/key.
I tried changing the source, but for some reason it doesn't accept just the file, only directories. 
Can you help me out with an example please? Thanks!

pschatzmann, 2022-07-14:
There are plenty of approaches:
- Some users use the AudioPlayer together with a setFileFilter() method call on the AudioSourceSdFat
- A leaner approach is just to start to copy the mp3 file to an EncodedAudioStream when you press the button.

pschatzmann, 2022-07-14:
I have never looked into this and honestly we don't have any influence on this. 
Have you tried with a smaller i2s buffer size but bigger buffer count ?

merugear, 2022-07-14:
woops, forgot to close with comment