pschatzmann/arduino-audiokit #48: My ESP32 Audio kit 2762 works with ES3833, not AC101

LieBtrau, 2022-11-16:
Hello,
Contrary to the info on [the main page](https://github.com/pschatzmann/arduino-audiokit) my ESP32 Audio Kit version 2762 works with the AUDIOKIT_BOARD=5.
As the audio codec is embedded the ESP32-A1S, not on the ESP32 Audio kit, it might be more correct to list the revision number of the ESP32-A1S instead of listing the revision number of the ESP32 Audio Kit.

The ESP32-A1S's revision is on the back of the module.  Mine is 2794.

pschatzmann, 2022-11-16:
Not sure if this is helpful: I don't think that pin assignments of the modules have changed between the revisions...

LieBtrau, 2022-11-17:
1. Apparently there are multiple versions of the module around : [Deeptronic web shop](https://www.deeptronic.com/store/zstock/blackstomp-core/esp32-a1s-ai-thinker-audio-module/).  These modules have a different codec, so I guess they're not firmware-compatible.
2. As you already mentioned on the main page of this project,  there are also different PCB versions of the ESP32 Audio Kit.

I agree with you that this is all very confusing.
My only purpose with this request was to indicate that my ESP32 Audio Kit 2762 doesn't work with the proposed AUDIOKIT_BOARD=6.  It works fine with AUDIOKIT_BOARD=5.

Aside from this, I've successfully run a few tests, such as using PlatformIO with the [web-radio example](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/streams-url_mp3-audiokit).

Thanks for the library and keep up the good work.

pschatzmann, 2022-11-17:
The README.md has been updated..

closed 2022-11-18 by pschatzmann