pschatzmann/arduino-audiokit #50: AI Thinker v2.2 3378

andrewwakeling, 2022-11-24:
I am able to get this board partially working.

The board appears to work with AUDIOKIT_BOARD values of 1 (lyrat_v4_3) or 7 (ai_thinker (ES8388) 2957).

I had to use AI_THINKER_ES8388_VOLUME_HACK, otherwise the volume levels are way too low (i.e. barely audible).

On earphones, I am only get the left audio (i.e. the right side is completely silent).

If anybody can share their solution for 3378 or point me in the rough direction to solve this, I would be grateful.

andrewwakeling, 2022-11-24:
I tried setting `AI_THINKER_ES8388_VOLUME_HACK` to 2 but it just made the volume very low again.

My main issue here is that I'm only getting audio from 1 channel (the left).

Here's my code:

```
#include <Arduino.h>
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "AudioCodecs/CodecFLAC.h"
#include "Password.h"

URLStream url(WIFI_SSID, WIFI_PASSWORD);
FLACDecoder dec;
AudioKitStream i2s;

void setup() {
  esp_wifi_set_ps(WIFI_PS_NONE);
  auto cfg = i2s.defaultConfig(TX_MODE);
  cfg.sample_rate = 44100;
  cfg.channels = 2;
  i2s.begin(cfg);
  i2s.setVolume(100);

  url.begin("http://192.168.1.1:8080/44100-stereo.flac");
  dec.setInputStream(url);
  dec.setOutputStream(i2s);
  dec.begin();
}

void loop() {
  dec.copy();
}
```

This currently isn't an urgent issue for me, but this might be good to resolve for others that have the 3378 board.

I also have 2762 board and the above code works perfectly.

owsis, 2023-07-27:
Try this, after `kit.begin();`
`es8388_write_reg(ES8388_DACCONTROL7, 0x1C);`