pschatzmann/arduino-audiokit #69: Changing MCLK Pin Default for ES8388

HosseinKasaei, 2023-02-24:
Hi,

I've made a custom board included ESP32 and ES8388. I've used another pin rather than GPIO0 for MCLK, but it doesn't work and I'm getting junk data. I tried to change the default MCLK pin in board_def.h and board_pins_config.c but it doesn't work. Do I have to stick to GPIO0 for MCLK?

pschatzmann, 2023-02-24:
To be honest I never looked in detail into this. I was just taking the logic which was implemented in the ADF.
I think by default the pin should be taken from the codec specific configuration file. You should have the possibility to provide your custom definition in the [AudioKitStreamConfig](https://pschatzmann.github.io/arduino-audio-tools/classaudio__tools_1_1_audio_kit_stream_config.html), but I never tested if this is properly working.

The ESP32 only supports 2 pins for the MCLK: GPIO0 and GPIO3.

But here is the key question: is this really used by the ES8388 ?

pschatzmann, 2023-02-24:
Just made a test and I confirm that the functionality of setting the masterclock pin is working as expected.
And I also confirm that this is needed by the ES8388: Setting it to 3 on an AudioKit was making that the audio did not work any more...

closed 2023-02-26 by pschatzmann