pschatzmann/arduino-audiokit #76: SD examples work with errors On Lyrat V4_3 board

timmcarson, 2023-04-07:
I successfully ran both sd_begin.ino and sdmmc_begin.ino on a Lyrat V4_3 board.  Is there a way to fix the gpio errors?

Thanks

Serial Monitor Output:

        rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
        configsip: 0, SPIWP:0xee
        clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
        mode:DIO, clock div:2
        load:0x3fff0030,len:1184
        load:0x40078000,len:13192
        load:0x40080400,len:3028
        entry 0x400805e4
        E (197) gpio: gpio_set_intr_type(148): GPIO interrupt type error
        [     9][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
        E (25) gpio: gpio_set_intr_type(148): GPIO interrupt type error
        [    34][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=18 scl=23 freq=100000
        Card Mount Success

pschatzmann, 2023-04-07:
Not sure where this is coming from. I am not calling gpio_set_intr_type() anywhere in my code!

timmcarson, 2023-04-07:
Hmm...  maybe something to do with Platformio?

pschatzmann, 2023-04-07:
I'm getting this in Arduino as well. 
Never had the time to look into it but everything is working w/o issues...

timmcarson, 2023-04-07:
thanks, I will ignore

pschatzmann, 2023-04-08:
I was cleanup up the API replacing all IDF calls with the proper Arduino ones.
This helped to resolve this issue

closed 2023-04-08 by pschatzmann