pschatzmann/arduino-audiokit #77: Power management of ES8388 using audio-kit lib

HosseinKasaei, 2023-04-13:
Hi,

I've designed a custom board including ES8388 and ESP32. My board is used for a BLE application and is supplied by a battery so the power management is very critical. I'm using audio-kit library and I want to be able to turn on ES8388 whenever is needed and turn it off most of the time to save power. Tried to use kit.end() but no luck. The current consumed in different situation is as follows :

1- kit is not configured in the main() : 48 mA
2- kit is configured in the main() and it's always turned on : 90 mA
3- kit is configured but I use kit.end() when a callback occurs and after that : 80 mA

I was expecting the current in 3 be the same as 1. Is it something about ESP32? maybe running kit.begin() is turning on something in it that kit.end() won't turn them off?

HosseinKasaei, 2023-04-14:
Thanks I tested these methods and the results were actually very good !
I have 20mA decrease when es8388_pa_power(false) and 5mA when Wire1.end() is called which left me with a current consumption of 55mA. Still can be improved because I have around 46mA when ES8388 is not configured at all. I'll appreciate any another idea and if I found by myself I'll share it here. Thanks again :)

pschatzmann, 2023-04-14:
So I suggest to set the logging to debug and check if es8388_deinit(void) is called. 
This should shut down the audio chip...

HosseinKasaei, 2023-04-14:
Here is the log. es8388_deinit is called : 

[image]

pschatzmann, 2023-04-14:
I just double checked the code, this should also shut down the Wire1.end(), so I guess the 5 mA are just measuring differences.

I will add the shut down of the power amplifer into the end() method.

Apart from this I don't see where the difference could come from...

pschatzmann, 2023-04-14:
Just another idea: if you do not use the SD drive make sure to deactivate it

HosseinKasaei, 2023-04-15:
> Just another idea: if you do not use the SD drive make sure to deactivate it

Yea I have deactivated it already

pschatzmann, 2023-04-23:
I am closing this issue because I don't see anything which could be wrong...

closed 2023-04-23 by pschatzmann