pschatzmann/arduino-audiokit #84: Have you had AUDIO_HAL_ADC_INPUT_LINE1 work?

Gaai, 2023-05-14:
I noticed in the input example you used AUDIO_HAL_ADC_INPUT_LINE2. I'm not sure how the two mics on the AudioKit are wired but is AUDIO_HAL_ADC_INPUT_LINE1 supposed to work as microphone input too? It doesn't for me. And I'm wondering if this is a defect on my board or pretty common.  

Line1 seems to be the default input AudioKitSettings.h:
`#define AUDIOKIT_DEFAULT_INPUT          AUDIO_HAL_ADC_INPUT_LINE1
`
And in AudioKitHAL.h:
`audio_hal_adc_input_t adc_input = AUDIOKIT_DEFAULT_INPUT; /*!<  set adc channel with audio_hal_adc_input_t*/
`

So I guess you had it working once.

Gaai, 2023-05-14:
Ah now I'm wondering if this might be the line input. I'll check right away.

pschatzmann, 2023-05-14:
All AudioKit boards have a hardware bug and you can't select between Aux in and microphone in.
https://www.pschatzmann.ch/home/2021/12/15/the-ai-thinker-audiokit-audio-input-bug/

closed 2023-05-14 by pschatzmann

Gaai, 2023-05-14:
Thanks a lot. Valuable info there that I hadn't fully read and taken in before!