pschatzmann/arduino-audiokit #88: Problem with base-i2s-a2dp

oloftangrot, 2023-05-23:
I have problem with getting the example to work. I can verify that the connection to the speaker work with bt_music_sender.ino. I have also verified that the I2S-microphone connection works with code from https://github.com/atomic14/esp32-i2s-mic-test, configured to use the same pins on my ESP32 DEVKITV1 as the base-i2s-a2dp example. Though, one thing that I had to change  was to connect the L/R pin to Vcc for I2S_CHANNEL_FMT_ONLY_LEFT which was the opposite with the wiring for the example. But I still don't get any sound in the speaker in the base-i2s-a2dp example. So it is possible that this indicates that the ESP-IDF driver somehow has switched L/R memory internally compared to some legacy version. But there are other differences with the esp32-i2s-mic-test seem to use the microphone in mono mode but the base-i2s-a2dp example uses a different mode, I2S_CHANNEL_FMT_RIGHT_LEFT, sampling rate is different and there might be other differences that I have not noticed.

I am open for suggestions on what to test but I currently don't understand the i2s or esp32 hardware or the code in base-i2s-a2dp to figure out myself where the problem can be.

pschatzmann, 2023-05-23:
You must be very confused: what is the connection to the AudioKit?

Don't open issues just because you are too lazy to read the documentation or the discussions opened by others.

I assume you use a INMP441

- https://github.com/pschatzmann/arduino-audio-tools/wiki/External-ADC 
- https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working

closed 2023-05-23 by pschatzmann

oloftangrot, 2023-05-24:
Confused and layzy, I can take more, you can do better that that.