pschatzmann/arduino-audiokit #94: ESP32-S2-Kaluga-1 v1.2 + ESP-LyraT-8311a v1.2 - Doesn't work

gonzoveliki, 2023-06-25:
1. It can't compile because ESP32-S2-Wrover module which is used on the Kaluga-1 v1. 2 board doesn't have two I2S, and compilation breaks at line 150 in audiokit_board.h
2. I've commented out lines 150-155, now it compiles ok, it talks via i2c on pins 7 and 8 with the codec 8311 successfully, but there is no sound from the generator output example.
3. PA_Enable is on pin10, it is enabled with kit.setSpeakerActive(true);

pschatzmann, 2023-06-25:
Since I don't have any of this board, I can't really test.
I found that the i2s masterclock was mapped to 0 instead of 35.
If my correction is still not working, I suggest that you double check the mapping of all pins...

gonzoveliki, 2023-06-25:
Thank you for your quick reply, indeed MCLK must be 35, and now I have output sound, but it is skipping and distorting... it seems like some I2S fifo buffer overrun...  I'm digging on it now, I will report when I catch where it comes from. Then I will do full tests with microphone input as well. Regards

gonzoveliki, 2023-06-25:
If I override default config's sample rate with
 "cfg.sample_rate = AUDIO_HAL_08K_SAMPLES;"
just before 
"kit.begin(cfg);"

pschatzmann, 2023-06-25:
Did you change the log level ? This would explain the reduced sample rate.
Not sure which setting is the microphone: did you try both cfg.adc_input = AUDIO_HAL_ADC_INPUT_LINE2 and cfg.adc_input = AUDIO_HAL_ADC_INPUT_LINE1

gonzoveliki, 2023-06-26:
Info:    bool audiokit::AudioKit::begin(audiokit::AudioKitConfig)
Info:    Selected board: 4
Info:    i2c_sda 8, i2c_scl: 7
Info:    SPI: cs: 13
Info:    clk: 14, miso: 2, mosi: 15
Info:    headphone detection not supported
Debug:   audio_hal_init
Debug:   audio_calloc: 1 * 36
Info:    i2c port: 0
Info:    i2c sda: 8
Info:    i2c scl: 7
Debug:   void* i2c_bus_create(i2c_port_t, i2c_config_t*)
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=1 value=48
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=2 value=0
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=3 value=16
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=22 value=36
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=4 value=16
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=5 value=0
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=11 value=0
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=12 value=0
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=16 value=31
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=17 value=127
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=0 value=128
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=0
Info:    ES8311 in Slave mode
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=2 value=0
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=5
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=5 value=0
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=3
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=3 value=16
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=4
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=4 value=16
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=7
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=7 value=0
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=8
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=8 value=255
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=6
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=6 value=3
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=1
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=1 value=63
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=6
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=6 value=3
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=19 value=16
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=27 value=10
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=28 value=106
Debug:   audio_codec_initialize -> 0
Info:    audio_codec_initialize-END-OK
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=9
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=10
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=9 value=12
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=10 value=12
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=9
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=10
Debug:   ES8311 in I2S Format
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=9 value=12
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=10 value=12
Debug:   SET: volume:51
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=50 value=51
Debug:   setupI2S
Info:    i2s bck_io_num: 18
Info:    i2s ws_io_num: 17
Info:    i2s data_out_num: 12
Info:    i2s data_in_num: 46
Info:    i2s_set_pin
Info:    - bck_io_num: 18
Info:    - ws_io_num: 17
Info:    - data_out_num: 12
Info:    - data_in_num: 46
Info:    - mck_io_num: 35
Debug:   audio_hal_ctrl_codec
Info:    Codec mode is 3, Ctrl:1
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=9
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=10
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=9 value=12
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=10 value=12
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=23 value=191
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=14 value=2
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=18 value=0
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=20 value=26
Debug:   i2c_bus_read_bytes: addr=48 reglen=1 datalen=1 - reg=20
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=20 value=26
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=13 value=1
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=21 value=64
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=55 value=72
Debug:   i2c_bus_write_bytes: addr=48 reglen=1 datalen=1 - reg=69 value=0
.....
Debug:   read: 1024
Debug:   write: 1024
Debug:   read: 1024
Debug:   write: 1024
.....

gonzoveliki, 2023-06-26:
> 

Yes, I did..same results, ADC reads samples full of harsh white noise, pumping up and down... Today I will dig deeper with an oscilloscope and Logic Analyzer to check I2S integrity. As I was able to trace the code flow of initialization all I2C registers were loaded properly as intended. I will make the same log for a LyraT v4.3 board which is working normally, to see are there some skipped sections of the init sequence.

pschatzmann, 2023-06-26:
I think the mic is connected as diff input: did you try AUDIO_HAL_ADC_INPUT_DIFFERENCE

gonzoveliki, 2023-06-26:
I did try with LINE1 and LINE2, now with DIFFERENCE... it's the same pumping noise up and down, but now i think there is a "difference" - the microphone is picking at least something, because when the noise is going down, I can hear something - if knock on the board or whistling straight into the mic. It looks that it comes from some kind of Auto Leveling Gain control which is out of lock.

gonzoveliki, 2023-06-26:
[image]

gonzoveliki, 2023-06-26:
This codec supports digital mics too (Digital Data goes like single ended mic into the chip, while mic's clock is sourced from I2S) 
Both for analog and digital mics it offers some basic DSP's gain, filtering and equalizer features on-board. I will fiddle with corresponding registers so hear is there any change.

pschatzmann, 2023-06-26:
Not sure if this helps, but I wondered about the following line:
#define ES8311_MCLK_SOURCE          1   /* 0 From MCLK, 1 From BCLK */
I would have expected this to be 0 ...

gonzoveliki, 2023-06-26:
I've tested that already, but I'm about to nail it. There is ALC control which needs to be constrained in gain, and there is muting levels to suppress noise.. These registers don't have proper factory defaults, so I fiddled randomly with values, and now it is recoding clean an undistorted audio. I will test thoroughly to see which are vital for proper initialization for recording, so you can add some extra functionality to HAL and chip driver. I think some conditional #ifdefs must be used as well, so the problem with missing I2S_NUM1 on S2-Wrover modules will be sorted.

gonzoveliki, 2023-06-27:
I'm still chasing the sampleRate issue, it is working good only with 8k. On all other possible rates it does skipping and clicking but sound pitch is unchanged - so it's not like sampling frequency mismatch, but more like I2S buffer overrun. Now I will connect the codec board LyraT to another MCU which is not S2, but WROVER-E, to check the possibility that issue comes from ESP core libs.

gonzoveliki, 2023-06-27:
Higher samplerates than 8k, consumes buffer qiucker, so if i increase buffer to 16000, and use 16k as sample rate, it sounds 1/2 of the time, if i make buffer 32k and sample rate also 32k, it sounds 1/4 of the second... I thought it is related with that this codec is mono, and SineGenerator feeds two 16bit integers for a sample into the buffer... but it is not that... I will dig more, so far I achieved good recording/playback audio @8k samplerate, after settled mic gain and ALC min/max levels...

pschatzmann, 2023-06-27:
Maybe it helps to compare the [actual 8311.c from ADF](https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver/es8311/es8311.c) with my old copy. I noticed that the coeff_div has changed.

gonzoveliki, 2023-06-27:
I have copied "coeff" table and most of the initialization code for latest ADF, but results were still the same, but now I'm almost 100% positive that problems is I2S related and not with the codec at all. It is improving if I manually compose  the "i2s_config_t i2s_config" forcing to mono format and different buffer_count and size scheme... somehow S2 chip clocks out unstable master and bit clocks, I can hear its jitter and frequency instability...

gonzoveliki, 2023-06-27:
That's good idea I will try tonight, meanwhile I have desoldered 4 2p caps C101,C32,C33,C34... and measuring with scope shows edges became a little bit more steep, and now it started to work with 11K sample rate, however a strange click is heard from time to time on random basis... that probably errors from shifted clock or a jitter accumulation...  it's still working perfect with 8k sr.
BTW: on other boards with codecs and ESP32 I saw resistors in series - like 22R or 33R on all I2S lines, but with Kaluga and LiraT (mini) there aren't such but those 22p caps probably intended to filter noise and smooth edges preventing overshoot/ringing...

gonzoveliki, 2023-06-27:
AudioKitHAL.h @line 640 
...    bool setupI2S(AudioKitConfig cnfg)...
but later cnfg is nowhere used - instead cfg which a local protected member in the class... was it intended like that? probably setupI2S must not be called with external parameters...

pschatzmann, 2023-06-27:
Actually I am not using this I2S output any more. 
In the AudioTools I rely on the I2S output functionality of the AudioTools and the AudioKit is just used to setup the codec

gonzoveliki, 2023-06-27:
Then I have to check how it is performing with AudioTools, so far with AudioKit it is working at 8k, and after I de-soldered few caps started to work @11k too. But it is not clean and sharp like 8388, sounds like de-tuned and with random clicks of an old vinyl record :)

pschatzmann, 2023-06-27:
I will remove the parameter for the setupI2S() method since this is not used and is confusing...

gonzoveliki, 2023-06-27:
okay... which example of AudioTools  should  I use for testing output and changing sample rate ?

pschatzmann, 2023-06-27:
I usually use https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/streams-generator-audiokit

gonzoveliki, 2023-06-27:
the Generator is sounding good, which is default AudioKitStream samplerate?!?
how to change/override SR?
Regards

gonzoveliki, 2023-06-27:
In fact kaluga's I2S_NUM is 0, because on LyraT-8311A v1.2 shield board there are routed two I2S pathways (for Codec-ES8311 and ADC1-ES7243), but the S2 chip which is on the base board have only one I2S controller built-in, so it's not possible to use 8311 and the ADC1 simultaneously, and I2S0 must be used with different pin mapping targeting ES8311 or ES7243,,,

gonzoveliki, 2023-06-27:
Confirmed, stream-input->output forwarding with AudioTools do work on 8311, but it still does need patching mic_gain and ALC max/min registers. Still digging on I2S issues with AudioKit, I'm looking now how it is done in AudioTools, and what could be the problem...

pschatzmann, 2023-06-27:
I found this: https://github.com/pschatzmann/arduino-audiokit/commit/d9cfb282169a06fa36e40bd803587be1471324d2

gonzoveliki, 2023-06-28:
I have already set interrupt flags prio to 0, and also added these at the bottom of the section.
#if ESP_IDF_VERSION_MAJOR >= 4 
            .fixed_mclk = 0, //(int) (cfg.fixed_mclk>0 ? cfg.fixed_mclk : 0 ),
            .mclk_multiple = I2S_MCLK_MULTIPLE_DEFAULT,
            .bits_per_chan = I2S_BITS_PER_CHAN_DEFAULT,
#endif            
Same results, all registers of codec are set properly, but with AudioTools via I2SESP.h it sounds good at any SR, while with build in read/write in AudioKit it does work only at 8k, and with some minor clicks and sparks at 11k... It's strange because it did improve a bit, when I have de-soldered edge smoothing caps from the I2S lines... but with AudioTools it outputs normal even caps are present.

gonzoveliki, 2023-06-28:
It seems like buffering issue, overrun at fifo buffers, or DMA... interrupt nesting... I will not stop until it is sorted, it's good to have the more simple AudioKit usable on all platforms, with supported boards and codecs... Please consider spitting board profile for Kaluga + ESP-LyraT-8311A v1.2 in two setups, one for ES8311 and another for ES7243...  so different pin mapping can be switched easy with the board selection choice

pschatzmann, 2023-06-28:
I would like to have this resolved as well, but currently I don't see any difference.

The idea is to use the I2SESP-New.h as soon Arduino supports the new API...

gonzoveliki, 2023-06-28:
This afternoon I will try to isolate what is causing these strange effects, for now its 100% that it has nothing to do with codec initialization, however some corrections would be necessary to setup the proper INPUT_LINE and mic gain + alc.
I will try to copy and use simplified version of the code from I2SESP, just one by one, hoping that at some moment problem will be resolved.. Its quite challenging, I love those strange situations, than both hardware and software has to fiddled to reach the root cause of the issue.

gonzoveliki, 2023-06-28:
You won't believed, but the cause of noises clicks and sound detune in pitch, is the SineWaveGenerator.h included in the output test... it is from the "float", i changed angle to double, and noises and clicks did stop immediately, however there is de-tune effect, but when i changed all floats to double, it sounds clean and nice at all sampleRates... because deltaTime is precomputed at setSampleRate... it lacks of resolution, and later when angle is accumulated, because it's not rounded at full cycle, floats do cause these distortion, and clicks at higher sampleRates... Unbelievable...

gonzoveliki, 2023-06-28:
i propose that as replacement the generator class of the output example

  int16_t readSample() {
    if ((m_cyc += m_frequency / sample_rate) > 1.0) m_cyc -= 1.0;
    return m_amplitude * sinf(double_Pi * m_cyc);
  }

gonzoveliki, 2023-06-28:
okay I will

gonzoveliki, 2023-06-28:
[output.zip](https://github.com/pschatzmann/arduino-audiokit/files/11895825/output.zip)
I'm sending a modified version of the output test for audiokit for preview.

gonzoveliki, 2023-06-28:
okay. i will put this example in a separate folder, you can just fix generator of the existing one.
now I'm working on input and microphone line and gain setups for ESP-LyraT-8311A board, and to enable ES7243 as well
in separate board profile

pschatzmann, 2023-06-28:
If your ESP32-S2-Kaluga-1 v1.2 changes are working, I propose to submit it. It is much easier to review some individual changes than many different things