pschatzmann/arduino-audiokit #95: input.ino output 0, 0 on S3-BOX-Lite

HonestQiao, 2023-07-03:
I have a S3-BOX-Lite.
[image]

Now ,I can output sound with [output.ino](https://github.com/pschatzmann/arduino-audiokit/blob/main/examples/output/output.ino), and play mp3.

But input.ino only output 0, 0.
[image]

What additional processing is required to record sound?

pschatzmann, 2023-07-04:
I did not add any functioality for thouse cases where a second codec adc ship needs to be set up.
However you can call the related functions yourself: ES7243

HonestQiao, 2023-07-05:
I will try it.

Get better soon!