pschatzmann/arduino-audiokit #101: Improve Input Quality

massa007, 2023-08-15:
Hi everyone,

I have setup the AI-Thinker Audio Kit (ES8388) with audiokit framework. I get an output and can also record vie On-Board mics.
The recording quality is quite limited in my case with a high frequency noise in the background and some lower noise. It is ok but far away from perfect!
I bought an external microphone which I can plug into the line in jack plug. 

1. Which Input do I have to set for using the audio jack line in? I do yurrently use `cfg.input_device = AUDIO_HAL_ADC_INPUT_LINE2;` and tried Line1, Line3 and ALL without success.
2. Any other options on how to improve recording quality?

pschatzmann, 2023-08-16:
If I remember right AUDIO_HAL_ADC_INPUT_LINE2 should be the microphones and AUDIO_HAL_ADC_INPUT_LINE1 is AUX in. Unfortunately in your version there is a hardware bug that prevents that you can deactivate the internal microphones

You can try to fine tune the input gain for your use case....

closed 2023-08-16 by pschatzmann

massa007, 2023-08-16:
Thanks a lot for your quick answer!
May I ask for any recommendation for a similar board without this bug/limitation and better recording quality?

pschatzmann, 2023-08-16:
The LyraT and the old ai_thinker (AC101)  work as designed!

I don't think that the quality will be better however. I am not sure if it is a problem with your board or just your expectations.
Maybe it's an issue with your power supply?