pschatzmann/arduino-audiokit #105: Board selection in arduino ide

hackintoshlover, 2023-09-21:
What board to select while uploading code using Arduino IDE?
There is no mention of esp32a1s.

closed 2023-09-21 by pschatzmann

pschatzmann, 2023-09-21:
I don't answer to questions that have already been answered before and are documented: It does not make any sense to give the same answer over and over again just because people are too lazy to read the documentation.

It's the first answer in the FAQ!

pschatzmann, 2023-09-21:
Here is just the reminder: https://github.com/pschatzmann/arduino-audiokit#support

hackintoshlover, 2023-09-21:
Guess you have ego issues. Since you can comment things without actually answering or asking.
Anyways.
The thing does not work and keeps on resetting, so i jasked just to clarify even after reading your wiki and blog, just to check if you could provide some useful inputs.
My board is definitely working since using it with vscode and esp-adf just works like charm.
Just thought to check it out with you repo over arduino.
Guess will stick to vs code.

pschatzmann, 2023-09-21:
No, it the lack of you being able to provide the requested information. esp32a1s boards have a regular ESP32.
If you have a custom solution which is not one of the commercial boards, you need to make sure that the custom pins are defined correctly