pschatzmann/arduino-audiokit #106: Output example will not compile if board type is set to 6

rogerclarkmelbourne, 2023-09-25:
### Problem Description

Output example will not compile if board type is set to 6

### Device Description

Not applicable   (AI Thinker / clone from AliExpress)

### Sketch

```cpp
N/A  using your example unmodified
```

### Other Steps to Reproduce

1. Open the Output example from this repo
2. Edit ArduinoKitSettings.h and  define   AUDIOKIT_BOARD  7
3. Observe

Compile error

Arduino\libraries\arduino-audiokit\src/audio_hal/audio_hal.h:185:25: error: conflicting types for 'AUDIO_CODEC_ES8388_DEFAULT_HANDLE'
 extern audio_hal_func_t AUDIO_CODEC_ES8388_DEFAULT_HANDLE;

### Provide your Version of the EP32 Arduino Core

2.0.13

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-09-25:
A correction has been committed.

rogerclarkmelbourne, 2023-09-25:
Thanks. It compiles OK now

closed 2023-09-25 by rogerclarkmelbourne