pschatzmann/arduino-audiokit #107: Example sd_begin does not compile

rogerclarkmelbourne, 2023-09-25:
### Problem Description

Example sd_begin does not compile

### Device Description

N/A (AI Thinker Audio kit)

### Sketch

```cpp
https://github.com/pschatzmann/arduino-audiokit/blob/main/examples/sd_begin/sd_begin.ino
```

### Other Steps to Reproduce

Attemp to compile the example

https://github.com/pschatzmann/arduino-audiokit/blob/main/examples/sd_begin/sd_begin.ino

See error message

Arduino\libraries\arduino-audiokit\examples\sd_begin\sd_begin.ino: In function 'void setup()':
sd_begin:20:22: error: 'class audiokit::AudioKit' has no member named 'pinSpiCs'
     if(!SD.begin(kit.pinSpiCs(), AUDIOKIT_SD_SPI)){

### Provide your Version of the EP32 Arduino Core

2.0.13

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

rogerclarkmelbourne, 2023-09-25:
BTW. Board type is set to 5, but this is irrelevant because I searched all files and the function pinSpiCs() does not exist
The only references are in the documentation

pschatzmann, 2023-09-25:
Oh, I actually never used this in the AudioTools, but relied on PIN_AUDIO_KIT_SD_CARD_CS.
The correction ha been committed...

pschatzmann, 2023-09-25:
I corrected the example to be in line with what I do in the AudioTools

rogerclarkmelbourne, 2023-09-25:
OK.

I've been having no success with the SD on this board from AliExpress

https://www.aliexpress.com/item/4000130915903.html

That example fails SD initialiation

`rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Initialization failed!
`

I also tried the mmc example, and that compiled but would not connect to the SD :-(

I have 2 of those boards, and SD does not worth on either of them. I tried multiple different SD cards

DIP switches are set to Off, On, On, Off, Off  
but I actually tried all combinations of DIP switch settings, all 32 and none works :-(

pschatzmann, 2023-09-25:
Double check the pin settings that are given in the Wiki

rogerclarkmelbourne, 2023-09-25:
OK. I'll check the pins.

pschatzmann, 2023-09-25:
Sometimes the issue is with the formatting. I recommend the official SD Formatter: https://www.sdcard.org/downloads/formatter/

rogerclarkmelbourne, 2023-09-25:
OK. I have feeling is a problem with pin mapping, but I'll try reformatting the card again

rogerclarkmelbourne, 2023-09-25:
I didn't mean your pin mapping was wrong, I suspect the boards I bought from AliExpress may be a clone of the AiThinker boards and may potentially be different :-(

pschatzmann, 2023-09-25:
I am getting an error as well: Just tried in the AudioTools, where this is the case as well, but the audio from the SD is playing

rogerclarkmelbourne, 2023-09-25:
Ah. OK.

Put a completely blank SD in at the moment to try to isolate the problem, so didn't bother listening for any audio

pschatzmann, 2023-09-25:
Oh,, stupid me: the SPI needs to be initialized with the proper pins

pschatzmann, 2023-09-25:
I extended the example to setup the pins.
This is however only necessary when not using calling kit.begin()

closed 2023-09-25 by pschatzmann

rogerclarkmelbourne, 2023-09-25:
Thanks.  Thats fixed the SD_begin example

rogerclarkmelbourne, 2023-09-25:
I presume all the examples need that line of code to be added.

i.e I just added it to your streams audiokit sd audiokit example and the card now works and I can record and playback the audio

pschatzmann, 2023-09-25:
No, this is automatically called with kit.begin(). In the example we are not doing that, so the SPI pins are not defined correctly

pschatzmann, 2023-09-25:
Just made a quick test and the following is working as expected:
```C++
#include "AudioKitHAL.h"

#include "SD.h"

AudioKit kit;

void setup(){
    Serial.begin(115200);
    
    // setup SPI with the coorrect pins: This is not necessary if you call kit.begin() with the SD active.
    //SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
    auto cfg = kit.defaultConfig(AudioOutput);
    kit.begin(cfg);

    // start SD
    if(!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS, AUDIOKIT_SD_SPI)){
        Serial.println("Card Mount Failed");
        return;
    } else {
        Serial.println("Card Mount Success");
    }
}

void loop() {

}
```

pschatzmann, 2023-09-25:
Can you point me to these examples ?

rogerclarkmelbourne, 2023-09-25:
I only saw that one

https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/streams-audiokit-sd-audiokit

BTW

the SDMMC_Begin example won't work for me

Same card works OK for normal SD 

```
E (56) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (56) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
Card Mount Failed
```

I set all DIP switches to on

Also. Unrelated..

But the player-sdfat-audiokit  example does not play some mp3's very well, they keep skipping and have strange noises in them.
I presumed this was because the data rate was too high (320 bps in some cases)

Hence why I stated to try the SDMMC examples becuase they should handle higher data rates from the SD card

pschatzmann, 2023-09-25:
Cool, that's the example that I have identified and corrected as well.
If you have decoding issues because of a high bitrate, you should increase the buffers as described in libhelix

rogerclarkmelbourne, 2023-09-25:
OK. Thanks

I presumed it was a SD data rate problem.

Actually I didn't realise that some of my MP3s had such high data rates. Most are 128kpbs or 192kpbs  , so I don't know why the first album I tried had 320bps ;-)

pschatzmann, 2023-09-25:
I was double checking the SDMMC as well and for me it is working.
```
18:40:51.546 -> configsip: 0, SPIWP:0xee
18:40:51.546 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:40:51.546 -> mode:DIO, clock div:1
18:40:51.546 -> load:0x3fff0030,len:1344
18:40:51.546 -> load:0x40078000,len:13964
18:40:51.546 -> load:0x40080400,len:3600
18:40:51.546 -> entry 0x400805f0
18:40:51.740 -> [    25][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
18:40:52.158 -> [   471][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
18:40:52.191 -> [   494][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=33 scl=32 freq=100000
18:40:52.288 -> Card Mount Success
```
Maybe you just forgot to disconnect the power for the settings to get activated. That's also the reason why I missed that the SD example was not working...

rogerclarkmelbourne, 2023-09-25:
Unfortunately, I still don't have any success with SD MMC.

I tried reformatting the card and power cycling the board, and removing and reinserting the card multiple times but I still get the same error

are you using your SDMMC_begin example, because your debug output looks a lot different to mine

pschatzmann, 2023-09-26:
I activated the Code Debug Level to Debug in the Tools menu, so that I could see any further messages from the core.

That's strange that this is not working: the SD pins are all the same on all boards, so I don't see any reason why this should not work. 

In order to switch off any negative impact from other pins, I suggest that you try the settings OFF, OFF, ON, OFF, OFF: This just leaves the CMD pin 15 active.

The SDMMC.begin command has additional parameter that you could try out...

pschatzmann, 2023-09-26:
With the settings OFF, ON, ON, OFF, OFF I even got the 4 bit mode to work!

rogerclarkmelbourne, 2023-09-26:
OK

Thanks

I'll give it another try.

Of course it could just be just that hardware is a non-exact clone of the AIThinker board.

pschatzmann, 2023-09-26:
Yes, each board is slightly different, but the SD pins are all the same across all ESP32 boards.

rogerclarkmelbourne, 2023-09-26:
I think one of the 2 boards I have is faulty.

I changed to the other board and the SD MMC example is now reporting that the card is working OK.

Strangely both boards are also now having problems with the USB device going on and off, over and over again.

I think I may abandon these clone audio dev boards and build from separate component modules.

pschatzmann, 2023-09-26:
That's maybe a good decision: Then you can also avoid the audio input bug that all AI Thinker boards have

rogerclarkmelbourne, 2023-09-26:
Thanks.

I have some PCM5102A boards. I'm not sure if I have anything else which is compatible with your libraries

BTW.  I also got the SD Fat example working in PlatformIO.

pschatzmann, 2023-09-26:
So what would your favorite microcontroller/audiochip combination be ?
I am struggling with the STM32 processors because there is no consistent I2S support in Arduino.

rogerclarkmelbourne, 2023-09-26:
Actually I was also considering using the RPi Pico. 
I do like the STM32 but the Pico seems to now be replacing the STM32F1 and STM32F4 boards, with its performance and price

pschatzmann, 2023-09-26:
PCB5102A is perfect if you just need audio output for all platforms that support i2s
STM32 and VS1053 might be a good and easy combination
The RP2040 should work for most of the scenarios - just keep in mind that it does not compare with the floating point performance of an ESP32

rogerclarkmelbourne, 2023-09-26:
Thanks

I didn't know that the RP2040 didn't have a FPU :-(

I just checked my box of modules and found I already have some VS1053 boards which I bought several years ago, so I may experiment with using them on a lower spec MCU e.g. STM32F103