pschatzmann/arduino-audiokit #111: LyraT 4.3 PA chip works with headphone plugged in

hhemmati81, 2023-12-06:
### Problem Description

I don't know if this falls under the audiokit repo or your audio-tools but I am having issues with the L/R speakers on the LyraT 4.3. They only activate when I plug a headphone in. Shouldn't this be the other way around?

### Device Description

LyraT 4.3

### Sketch

```cpp
Using player-sd_a2dp-audiokit example from audio-tools
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-12-07:
In Ajludiokit.h you should find the line in the begin method, that activates the speaker. By default the output should go to both the speaker and to the earphones

hhemmati81, 2023-12-08:
After reviewing it a bit further it looks like your code is working properly but something is wrong with the way LyraT is handling the audio. The L channel has about half the volume and the R channel has about maybe 10% audio. Now if I short out the pins on the headphone jack I get full audio from both L and R. The amp seems to be software-controlled but something is definitely not right about it unless I have a defective board.

pschatzmann, 2023-12-09:
It might be a power issue? 

Did you try to select a dac_output value, that would use not both lines?

hhemmati81, 2023-12-10:
I think my issue is related to this [https://esp32.com/viewtopic.php?t=12407](url)  since I am using the Line-in for audio input. I haven't tried to remove the caps as people suggested but I added a couple of 4ohms to short out the headphone jack L/R. It has solved the audio level issue for me and has significantly improved the sound quality as well.

I haven't tried using A2DP as a source yet to see if it is going to work with that as well since I haven't been able to figure out how to make it work with WiFi. 

I have to have WiFi on since that's how I communicate with my app and the A2DP that comes with the audio-tools causes a core panic. I used to use the standalone A2DP package before I moved to the LyraT and I didn't have any issues with that one running together with WiFi.